jsk-ros-pkg/jsk_control #201: footcoordsの接地判定で力センサ値を座標変換する

mmurooka, 2015-01-22:
@garaemon

https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_footstep_controller/src/footcoords.cpp
の接地判定で，/lfsensor, /rfsensor をそのまま読んでいますが，
力センサ座標系に変換する必要がありそうです．

HRP2とSTAROで力センサの座標系が異なっていました．
（`rostopic echo /lfsensor`で，HRP2は接地時にZ成分が正の値，STAROは接地時にZ成分が負の値になる．）
Rviz上の力センサを表す矢印は，どちらのロボットも正しく出ているのでtfで変換すれば正しくなると思います．

garaemon, 2015-01-22:
STAROとHRP2の力センサ座標系ってなんだろう? frame教えてもらえる?
おそらく、STAROは

`hoge->fuga->wrench`みたいになんか一個挟まってて、その一つ上が知りたいです. たぶんhogeとfugaが反対を向いているんじゃないかな

garaemon, 2015-01-22:
STAROの力センサ、回転だけじゃなくて位置も変えてるのか,,, 面倒ですねー

mmurooka, 2015-01-22:
STAROの足力センサまわりのtfです．
HRP2と同じに見えます，というので合っていますでしょうか．

[image]

garaemon, 2015-01-22:
https://github.com/jsk-ros-pkg/jsk_control/issues/202
これで対応しました。
とりあえずlleg_end_coords/rleg_end_coordsと同じ座標系で値を見るようにしましたが、`lfoot_sensor_frame`, `rfoot_sensor_frame`を適切に指定してください

garaemon, 2015-01-22:
なるほど、このプルリクエストで動くようになると思います

snozawa, 2015-01-22:
ちなみに接地判定はどういう処理をしていますか？

hrpsys-baseの方でも似たような値を計算してますが、取得できるとよかったりしますでしょうか？

garaemon, 2015-01-22:
lowpass filterかけて、閾値処理ですね。
取得できると良いですね、という話は以前も出てましたが。

garaemon, 2015-01-22:
ついでに、ですがhrpsysではエンコーダからルートリンクの速度とかは計算していますか? 
それも欲しいですね

mmurooka, 2015-01-22:
ありがとうございます．

garaemon, 2015-01-22:
マージしたので確認してみてください

mmurooka, 2015-01-22:
```
    pnh.param("lfoot_sensor_frame", lfoot_sensor_frame_, std::string("lleg_end_coords"));
    pnh.param("rfoot_sensor_frame", rfoot_sensor_frame_, std::string("rleg_end_coords"));
```
だと以下のエラーでした．
```
terminate called after throwing an instance of 'tf2::LookupException'
  what():  "lleg_end_coords" passed to lookupTransform argument source_frame does not exist.
[footcoords-28] process has died [pid 15889, exit code -6, cmd /home/leus/ros/hydro/devel/lib/jsk_fo
otstep_controller/footcoords __name:=fo
```

```
    pnh.param("lfoot_sensor_frame", lfoot_sensor_frame_, std::string("lfsensor"));
    pnh.param("rfoot_sensor_frame", rfoot_sensor_frame_, std::string("rfsensor"));
```
では，エラーが出ませんでしたが，座標系が違うままでした．

garaemon, 2015-01-22:
ふむ、一回目は失敗するのかなぁ、対応するのでちょい待ちです

mmurooka, 2015-01-22:
あとから，
> rosrun jsk_footstep_controller footcoords

したら，上がって座標も合っていて解決しました．

start_ros_bridgeでいろいろ一緒に上がるなかで，
lleg_end_coords のtfがpublishされる前に，footcoordsが始まってしまうのが問題みたいですね．

mmurooka, 2015-01-22:
> あとから，
>>   rosrun jsk_footstep_controller footcoords

したら

のときも，
5回に3回くらいは，
```
terminate called after throwing an instance of 'tf2::LookupException'
  what():  "lleg_end_coords" passed to lookupTransform argument source_frame does not exist. 
Aborted (core dumped)
```
で落ちてしまいます．
5回に2回の上がるときは，
```
[ERROR] [1421902329.922831239]: transform error: Lookup would require extrapolation into the future.  Requested time 1421902329.922467916 but the latest data is at time 1421902329.877795528, when looking up transform from frame [lleg_end_coords] to frame [lfsensor]
[ERROR] [1421902329.922878461]: failed to resolve tf of force sensor
```
というエラーが5秒くらい出てその後は出なくなって，正しい接地状態が得られます．

garaemon, 2015-01-22:
了解です、五分待ってください

✉︎ Ryohei

garaemon, 2015-01-22:
https://github.com/jsk-ros-pkg/jsk_control/pull/203
これでどうだ?

garaemon, 2015-01-22:
マージしたので試してみてください。今度こそうまくいくはず

mmurooka, 2015-01-22:
動きました．ありがとうございました．

closed 2015-01-22 by mmurooka