jsk-ros-pkg/jsk_control #211: eus_nlopt cannot compile with `catkin build`

garaemon, 2015-02-05:
https://github.com/jsk-ros-pkg/jsk_common/pull/736

```
$ catkin b eus_nlopt -i -v
==> Expanding alias 'b' from 'catkin b eus_nlopt -i -v' to 'catkin build eus_nlopt -i -v'
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/hrpuser/ros/hydro
Source Space:       [exists] /home/hrpuser/ros/hydro/src
Build Space:        [exists] /home/hrpuser/ros/hydro/build
Devel Space:        [exists] /home/hrpuser/ros/hydro/devel
Install Space:     [missing] /home/hrpuser/ros/hydro/install
DESTDIR:                     None
------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------ 
Found '209' packages in 0.4 seconds. 
Starting ==> nlopt                                                                                                                                                                                                                                             
[nlopt] ==> '/home/hrpuser/ros/hydro/build/nlopt/build_env.sh /usr/bin/make cmake_check_build_system' in '/home/hrpuser/ros/hydro/build/nlopt'                                                                                                                 
[nlopt] <== '/home/hrpuser/ros/hydro/build/nlopt/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'                                                                                                                            
[nlopt] ==> '/home/hrpuser/ros/hydro/build/nlopt/build_env.sh /usr/bin/make -j4 -l4' in '/home/hrpuser/ros/hydro/build/nlopt'                                                                                                                                  
[nlopt] <== '/home/hrpuser/ros/hydro/build/nlopt/build_env.sh /usr/bin/make -j4 -l4' finished with return code '0'                                                                                                                                             
Finished <== nlopt     [ 0.5 seconds ]                                                                                                                                                                                                                         
Starting ==> eus_nlopt                                                                                                                                                                                                                                         
[eus_nlopt] ==> '/home/hrpuser/ros/hydro/build/eus_nlopt/build_env.sh /usr/bin/cmake /home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_control/eus_nlopt -DCATKIN_DEVEL_PREFIX=/home/hrpuser/ros/hydro/devel -DCMAKE_INSTALL_PREFIX=/home/hrpuser/ros/hydro/install' in '/home/hrpuser/ros/hydro/build/eus_nlopt' 
[eus_nlopt]   eus_nlopt: use catkin_make                                                                                                                                                                                                                       
[eus_nlopt] -- Using CATKIN_DEVEL_PREFIX: /home/hrpuser/ros/hydro/devel                                                                                                                                                                                        
[eus_nlopt] -- Using CMAKE_PREFIX_PATH: /home/hrpuser/ros/hydro/devel;/opt/ros/hydro                                                                                                                                                                           
[eus_nlopt] -- This workspace overlays: /home/hrpuser/ros/hydro/devel;/opt/ros/hydro                                                                                                                                                                           
[eus_nlopt] -- Using PYTHON_EXECUTABLE: /usr/bin/python                                                                                                                                                                                                        
[eus_nlopt] -- Python version: 2.7                                                                                                                                                                                                                             
[eus_nlopt] -- Using Debian Python package layout                                                                                                                                                                                                              
[eus_nlopt] -- Using CATKIN_ENABLE_TESTING: ON                                                                                                                                                                                                                 
[eus_nlopt] -- Call enable_testing()                                                                                                                                                                                                                           
[eus_nlopt] -- Using CATKIN_TEST_RESULTS_DIR: /home/hrpuser/ros/hydro/build/eus_nlopt/test_results                                                                                                                                                             
[eus_nlopt] -- Found gtest sources under '/usr/src/gtest': gtests will be built                                                                                                                                                                                
[eus_nlopt] -- catkin 0.5.86                                                                                                                                                                                                                                   
[eus_nlopt] CMake Error at /home/hrpuser/ros/hydro/devel/share/nlopt/cmake/nloptConfig.cmake:141 (message):                                                                                                                                                    
[eus_nlopt]   Project 'eus_nlopt' tried to find library 'nlopt_cxx'.  The library is                                                                                                                                                                           
[eus_nlopt]   neither a target nor built/installed properly.  Did you compile project                                                                                                                                                                          
[eus_nlopt]   'nlopt'? Did you find_package() it before the subdirectory containing its                                                                                                                                                                        
[eus_nlopt]   code is included?                                                                                                                                                                                                                                
[eus_nlopt] Call Stack (most recent call first):                                                                                                                                                                                                               
[eus_nlopt]   /opt/ros/hydro/share/catkin/cmake/catkinConfig.cmake:72 (find_package)                                                                                                                                                                           
[eus_nlopt]   catkin.cmake:5 (find_package)                                                                                                                                                                                                                    
[... cut ...]                                        

    Failed to build package 'eus_nlopt' because the following command:

        # Command run in directory: /home/hrpuser/ros/hydro/build/eus_nlopt
        /home/hrpuser/ros/hydro/build/eus_nlopt/build_env.sh /usr/bin/cmake /home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_control/eus_nlopt -DCATKIN_DEVEL_PREFIX=/home/hrpuser/ros/hydro/devel -DCMAKE_INSTALL_PREFIX=/home/hrpuser/ros/hydro/install

    Exited with return code: 1 
[build] Runtime: 3.2 seconds 
```

k-okada, 2015-02-05:
do you have this error after https://github.com/jsk-ros-pkg/jsk_common/pull/736?

garaemon, 2015-02-05:
Yes I had

garaemon, 2015-02-06:
fixed by https://github.com/jsk-ros-pkg/jsk_common/issues/738

garaemon, 2015-02-06:
Still failed

`catkin build eus_nlopt` => success
`catkin build` => failed on eus_nlopt