jsk-ros-pkg/jsk_control #289: eus_qp could not compile

iory, 2015-04-24:
```
[eus_qp] ==> '/home/iori/ros/hydro/build/eus_qp/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/iori/ros/hydro/build/eus_qp'
[100%] Building CXX object CMakeFiles/eus_qp.dir/src/qp_lib.cpp.o
In file included from /usr/include/eigen3/Eigen/Core:277:0,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /home/iori/ros/hydro/src/jsk-ros-pkg/jsk_control/eus_qp/src/qp_lib.cpp:3:
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of class Eigen::DenseBase<Eigen::Matrix<float, 4, 1> >:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from class Eigen::MatrixBase<Eigen::Matrix<float, 4, 1> >
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:65:7:   required from class Eigen::PlainObjectBase<Eigen::Matrix<float, 4, 1> >
/usr/include/eigen3/Eigen/src/Core/Matrix.h:140:7:   required from class Eigen::Matrix<float, 4, 1>
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:287:16:   required from class Eigen::Quaternion<float>
/usr/include/eigen3/Eigen/src/Geometry/arch/Geometry_SSE.h:35:3:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: Eigen::internal::size_at_compile_time<4, 1>::<anonymous enum> vs Eigen::internal::traits<Eigen::Matrix<float, 4, 1> >::<anonymous enum> [-Wenum-compare]
       InnerSizeAtCompileTime = int(IsVectorAtCompileTime) ? SizeAtCompileTime
                                                           ^
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of class Eigen::DenseBase<Eigen::Matrix<double, 4, 1> >:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from class Eigen::MatrixBase<Eigen::Matrix<double, 4, 1> >
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:65:7:   required from class Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 1> >
/usr/include/eigen3/Eigen/src/Core/Matrix.h:140:7:   required from class Eigen::Matrix<double, 4, 1>
/usr/include/eigen3/Eigen/src/Geometry/Quaternion.h:287:16:   required from class Eigen::Quaternion<double>
/usr/include/eigen3/Eigen/src/Geometry/arch/Geometry_SSE.h:76:3:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: Eigen::internal::size_at_compile_time<4, 1>::<anonymous enum> vs Eigen::internal::traits<Eigen::Matrix<double, 4, 1> >::<anonymous enum> [-Wenum-compare]
In file included from /usr/include/boost/math/special_functions/round.hpp:13:0,
                 from /opt/ros/hydro/include/ros/time.h:58,
                 from /opt/ros/hydro/include/ros/console.h:39,
                 from /opt/ros/hydro/include/ros/assert.h:35,
                 from /opt/ros/hydro/include/ros/common.h:36,
                 from /home/iori/ros/hydro/src/jsk-ros-pkg/jsk_control/eus_qp/src/qp_lib.cpp:6:
/usr/include/boost/math/tools/config.hpp:277:7: error: fexcept_t does not name a type
       fexcept_t m_flags;
       ^
/usr/include/boost/math/tools/config.hpp: In constructor boost::math::detail::fpu_guard::fpu_guard():
/usr/include/boost/math/tools/config.hpp:269:27: error: m_flags was not declared in this scope
          fegetexceptflag(&m_flags, FE_ALL_EXCEPT);
```

@s-noda

s-noda, 2015-04-24:
I could have built this package 3 days ago by using newly installed Ubuntu 12.04 machine and hydro ROS environment. But, I will check again soon. 

Is your environment up to date, @iory ?

In my environment, ```fexcept_t``` is seemed to be declared in ```bits/fenv.h```.
```bits/fenv.h``` is included in ```boost/detail/fenv.hpp```  if the following condition was true;
```
#if defined(__clang__)       &&                       \
    defined(__GNU_LIBRARY__) && /* up to version 5 */ \
    defined(__GLIBC__)          /* version 6 + */
```

And this ```boost/detail/fenv.hpp```  file is included in ```boost/math/tools/config.hpp``` just above your error line.

So, this problem may result from your compiler settings.
What kinds of compiler did you use?

If this is true, by adding the following line; 
``` include <bits/fenv.h> ```
at the top line of qp_lib.cpp, you may be able to build eus_qp.

garaemon, 2015-12-09:
solved?

snozawa, 2016-04-08:
I'll close this PR.
If you have same error, please reopen this.

closed 2016-04-08 by snozawa