jsk-ros-pkg/jsk_control #349: eus_qpoases not installing headers and libs

airballking, 2015-06-19:
Hi everyone,

it seems your (catkin) CMakeLists of eus_qpoases does not install any source files or build artifacts. Is the intended behavior? 

My use-case: I want to depend on the C++ headers and libs of qpoases through depending on your ROS package. However, I cannot find any of the h-files in the debians. My system: Ubuntu 14.04, ROS Indigo, all updated on June, 18th, 2015.

Best,
Georg.

P.S.:  Cool job wrapping this solver for euslisp! I wish we had enough man-power to pursue similar efforts in roslisp...

s-noda, 2015-06-19:
Hi @airballking, thanks to use our package.

You intend to use qpOASES's header and library files such as qpOASES.hpp and libqpOASES.so?

These files are created in 
${CMAKE_CURRENT_BINARY_DIR}/build/qpOASES/libs
and
${CMAKE_CURRENT_BINARY_DIR}/build/qpOASES-source/include.

So,  your object might be achieved through adding the following lines at the bottom of catkin.cmake file.
```
install(DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}/build/qpOASES/libs/
  DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION})
install(DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}/build/qpOASES-source/include/
  DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION})
```

Could you check if it's ok for you?
Besides, give us pull request if possible?

airballking, 2015-06-19:
Hi @s-noda,

thanks for the very fast answer and proposition! I tried it out, and it works fine on my local pc. Both the header-files and libs show up in the install-space of my catkin-workspace.

I created a pull request which references this issue.

Best,
Georg.

P.S.: Again, thanks for wrapping this lib for ROS. This saved me some time, today. :)

k-okada, 2015-06-20:
closed via #350

closed 2015-06-20 by k-okada