jsk-ros-pkg/jsk_control #424: [jsk_footstep_planner] Empty Frame error has occured in heightmap.launch with PR2

aginika, 2015-08-10:
I'm using PR2 hydro and jsk_footstep_planner's heightmap.launch.
Simply I got below error.
```
terminate called after throwing an instance of 'tf2::InvalidArgumentException'
  what():  Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty
```

I wrote below launch file.

```
<launch>
  <arg name="input_cloud" default="/tilt_scan_shadow_filtered"/>
  <!-- <arg name="input_cloud" default="/kinect_head_c2/depth_registered/points"/> -->

  <!-- Too remove Ceiling Plane -->
  <node pkg="nodelet" type="nodelet" name="pass_through_z"
      args="standalone pcl/PassThrough">
  <remap from="~input" to="$(arg input_cloud)" />
    <rosparam>
      filter_field_name: z 
      filter_limit_min: 0
      filter_limit_max: 1.2
    </rosparam>
  </node>

  <include file="$(find jsk_footstep_planner)/launch/heightmap_perception/heightmap.launch">
    <arg name="STAND_FRAME" value="map"/>
    <arg name="ROBOT_FRAME" value="base_footprint"/>
    <arg name="DEBUG_VIEW" default="false" />
    <arg name="INPUT" value="pass_through_z/output"/>
  </include>
</launch>
```

/pass_through_z/output is publishing well and we can confirm it on rviz.
And theirs echo is as below.
```
$ rostopic echo /tilt_scan_shadow_filtered --noarr 
header: 
  seq: 297901
  stamp: 
    secs: 1439205155
    nsecs: 999429120
  frame_id: base_footprint
height: 1
width: 165
is_bigendian: False
point_step: 16
row_step: 2640
is_dense: True
---
```

What is the problem of the tf2 empty frame error?
And below is error.
```
$ roslaunch jsk_pr2_startup heightmap.launch 
... logging to /home/inagaki/.ros/log/089a57d2-3f3f-11e5-987a-001517ebc305/roslaunch-pr1040-13749.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pr1040:34790/

SUMMARY
========

PARAMETERS
 * /accumulated_heightmap/center_frame_id
 * /accumulated_heightmap/fixed_frame_id
 * /accumulated_heightmap_pointcloud_static/target_frame_id
 * /heightmap_nodelet/nodelets
 * /latest_heightmap/max_x
 * /latest_heightmap/max_y
 * /latest_heightmap/min_x
 * /latest_heightmap/min_y
 * /latest_heightmap/resolution_x
 * /latest_heightmap/resolution_y
 * /map_origin_polygon/frame_ids
 * /map_origin_polygon/polygon_array
 * /map_origin_polygon/use_message
 * /pass_through_z/filter_field_name
 * /pass_through_z/filter_limit_max
 * /pass_through_z/filter_limit_min
 * /robot_center_map_origin_points/target_frame_id
 * /robot_center_pointcloud/target_frame_id
 * /robot_center_pointcloud_with_map_origin_points/input_topics
 * /robot_center_pointcloud_with_map_origin_points/max_queue_size
 * /robot_center_pointcloud_with_map_origin_points/output_frame
 * /rosdistro
 * /rosversion

NODES
  /heightmap_visualization/latest/
    colorize_heightmap (jsk_perception/colorize_float_image)
  /heightmap_visualization/accumulated/
    colorize_heightmap (jsk_perception/colorize_float_image)
  /heightmap_visualization/filtered/
    colorize_heightmap (jsk_perception/colorize_float_image)
  /
    heightmap_nodelet (jsk_topic_tools/standalone_complexed_nodelet)
    pass_through_z (nodelet/nodelet)

ROS_MASTER_URI=http://c1:11311

core service [/rosout] found
process[pass_through_z-1]: started with pid [14094]
process[heightmap_nodelet-2]: started with pid [14226]
[ INFO] [1439205071.994113391]: [int main(int, char**)] i_nodelet 0
[ INFO] [1439205071.994286476]: [int main(int, char**)] remapping: /map_origin_polygon/input => /robot_center_pointcloud/output
[ INFO] [1439205072.380613897]: [int main(int, char**)] Succeeded to load nodelet [/map_origin_polygon -- jsk_pcl/StaticPolygonArrayPublisher]
[ INFO] [1439205072.380673164]: [int main(int, char**)] i_nodelet 1
[ INFO] [1439205072.380781669]: [int main(int, char**)] remapping: /map_origin_polygon_sampler/input/polygons => /map_origin_polygon/output_polygons
[ INFO] [1439205072.380853613]: [int main(int, char**)] remapping: /map_origin_polygon_sampler/input/coefficients => /map_origin_polygon/output_coefficients
[ INFO] [1439205072.413210877]: [int main(int, char**)] Succeeded to load nodelet [/map_origin_polygon_sampler -- jsk_pcl/PolygonPointsSampler]
[ INFO] [1439205072.413271919]: [int main(int, char**)] i_nodelet 2
[ INFO] [1439205072.413344045]: [int main(int, char**)] remapping: /robot_center_map_origin_points/input => /map_origin_polygon_sampler/output_xyz
process[heightmap_visualization/latest/colorize_heightmap-3]: started with pid [14346]
[ INFO] [1439205072.724251534]: [static tf::TransformListener* jsk_pcl_ros::TfListenerSingleton::getInstance()] instantiating tf::TransformListener
[ INFO] [1439205072.732965772]: [int main(int, char**)] Succeeded to load nodelet [/robot_center_map_origin_points -- jsk_pcl/TfTransformCloud]
[ INFO] [1439205072.733217773]: [int main(int, char**)] i_nodelet 3
[ INFO] [1439205072.733428916]: [int main(int, char**)] remapping: /robot_center_pointcloud/input => /pass_through_z/output
[ INFO] [1439205072.742064220]: [int main(int, char**)] Succeeded to load nodelet [/robot_center_pointcloud -- jsk_pcl/TfTransformCloud]
[ INFO] [1439205072.742124975]: [int main(int, char**)] i_nodelet 4
[ INFO] [1439205072.881342730]: [onInit] Subscribing to 2 user given topics as inputs:
[ INFO] [1439205072.881438975]:  - /robot_center_pointcloud/output
[ INFO] [1439205072.881481090]:  - /robot_center_map_origin_points/output
[ INFO] [1439205072.888799222]: [int main(int, char**)] Succeeded to load nodelet [/robot_center_pointcloud_with_map_origin_points -- pcl/PointCloudConcatenateDataSynchronizer]
[ INFO] [1439205072.888858916]: [int main(int, char**)] i_nodelet 5
[ INFO] [1439205072.889366658]: [int main(int, char**)] remapping: /latest_heightmap/input => /robot_center_pointcloud_with_map_origin_points/output
[ INFO] [1439205072.938153360]: [int main(int, char**)] Succeeded to load nodelet [/latest_heightmap -- jsk_pcl/HeightmapConverter]
[ INFO] [1439205072.938215729]: [int main(int, char**)] i_nodelet 6
[ INFO] [1439205072.938309861]: [int main(int, char**)] remapping: /latest_complete_heightmap/input => /latest_heightmap/output
[ INFO] [1439205072.967627914]: [int main(int, char**)] Succeeded to load nodelet [/latest_complete_heightmap -- jsk_pcl/HeightmapMorphologicalFiltering]
[ INFO] [1439205072.967688307]: [int main(int, char**)] i_nodelet 7
[ INFO] [1439205072.967777682]: [int main(int, char**)] remapping: /accumulated_heightmap/input => /latest_complete_heightmap/output
[ INFO] [1439205072.967855700]: [int main(int, char**)] remapping: /accumulated_heightmap/input/prev_pointcloud => /accumulated_heightmap_pointcloud/output
[ INFO] [1439205072.988036257]: [int main(int, char**)] Succeeded to load nodelet [/accumulated_heightmap -- jsk_pcl/HeightmapTimeAccumulation]
[ INFO] [1439205072.988104125]: [int main(int, char**)] i_nodelet 8
[ INFO] [1439205072.988206560]: [int main(int, char**)] remapping: /accumulated_heightmap_pointcloud/input => /accumulated_heightmap/output
[ INFO] [1439205073.000327509]: [int main(int, char**)] Succeeded to load nodelet [/accumulated_heightmap_pointcloud -- jsk_pcl/HeightmapToPointCloud]
[ INFO] [1439205073.000400186]: [int main(int, char**)] i_nodelet 9
[ INFO] [1439205073.000562136]: [int main(int, char**)] remapping: /accumulated_heightmap_pointcloud_static/input => /accumulated_heightmap_pointcloud/output
[ INFO] [1439205073.009599123]: [int main(int, char**)] Succeeded to load nodelet [/accumulated_heightmap_pointcloud_static -- jsk_pcl/TfTransformCloud]
process[heightmap_visualization/filtered/colorize_heightmap-4]: started with pid [14553]
terminate called after throwing an instance of 'tf2::InvalidArgumentException'
  what():  Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty
[heightmap_nodelet-2] process has died [pid 14226, exit code -6, cmd /home/inagaki/ros/hydro/devel/lib/jsk_topic_tools/standalone_complexed_nodelet __name:=heightmap_nodelet __log:=/home/inagaki/.ros/log/089a57d2-3f3f-11e5-987a-001517ebc305/heightmap_nodelet-2.log].
log file: /home/inagaki/.ros/log/089a57d2-3f3f-11e5-987a-001517ebc305/heightmap_nodelet-2*.log

```

garaemon, 2015-08-10:
Are map and base_footprint available?

garaemon, 2015-08-10:
Run with gdb to know which thread is dead by tf2 exception

aginika, 2015-08-10:
@garaemon When I executed on my local, it works well (For try to do gdb). Thank you. Might be my mistake .

closed 2015-08-10 by aginika