jsk-ros-pkg/jsk_control #457: jsk_footstep_planner　footstep-planner-node.lがセグフォで落ちます

YuOhara, 2015-09-14:
gdbの出力です。
```
http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from bash...(no debugging symbols found)...done.
(gdb) r
Starting program: /bin/bash /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_roseus/roseus/bin/roseus /home/leus/ros/indigo/src/jsk-ros-pkg/jsk_control/jsk_footstep_planner/euslisp/footstep-planner-node.l planes:=/ocs/env_server/occlusion_result_polygons grid_maps:=/ocs/env_server/output footstep_planner/result:=footstep_planner/result_non_refined __name:=footstep_planner __log:=/home/leus/.ros/log/f3764420-5ad1-11e5-a3f2-10c37b94cec8/footstep_planner-1.log
process 20427 is executing new program: /home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/eusgl
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
configuring by "/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x47e9030[16374] --> 0x4c47d00[32748] top=3c4d
irtgl irtviewer 
EusLisp 9.15(0be187a 1.0.10) for Linux64 created on leus-All-Series(Mon Sep 7 21:32:57 JST 2015)
roseus ;; loading roseus("22cb71e") on euslisp((9.15 leus-All-Series Mon Sep 7 21:32:57 JST 2015 0be187a 1.0.10))
[ INFO] [1442236907.425611128]: install ros::roseus-sigint-handler
eustf roseus_c_util whr[New Thread 0x7fffe4565700 (LWP 20436)]
[New Thread 0x7fffe3d64700 (LWP 20437)]
e[New Thread 0x7fffe3563700 (LWP 20438)]
[New Thread 0x7fffe2d62700 (LWP 20450)]
[New Thread 0x7fffe2561700 (LWP 20465)]
[New Thread 0x7fffe1a80700 (LWP 20526)]
[Thread 0x7fffe2561700 (LWP 20465) exited]
[ INFO] [1442236911.251316899, 1432.896000000]: Updating grid model
[ INFO] [1442236911.287036150, 1432.909000000]: :init takes 0.035449 secs (0.035449 sec per trial, 1 trials)
[ INFO] [1442236911.396101553, 1432.957000000]: :init takes 0.108868 secs (0.108868 sec per trial, 1 trials)
[ INFO] [1442236911.449724234, 1432.969000000]: :init takes 0.053239 secs (0.053239 sec per trial, 1 trials)
[ INFO] [1442236911.453694005, 1432.969000000]: :init takes 0.003699 secs (0.003699 sec per trial, 1 trials)

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe4565700 (LWP 20436)]
0x00007fffebbc4063 in boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::signal0_impl<void, boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::slot_invoker, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot0<void, boost::function<void ()> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot0<void, boost::function<void ()> >, boost::signals2::mutex> >::lock_next_callable() const ()
   from /opt/ros/indigo/lib/libroscpp.so
(gdb) where
#0  0x00007fffebbc4063 in boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::signal0_impl<void, boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::slot_invoker, std::_List_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot0<void, boost::function<void ()> >, boost::signals2::mutex> > >, boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot0<void, boost::function<void ()> >, boost::signals2::mutex> >::lock_next_callable() const ()
   from /opt/ros/indigo/lib/libroscpp.so
#1  0x00007fffebbc47a3 in boost::signals2::detail::signal0_impl<void, boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::operator()() () from /opt/ros/indigo/lib/libroscpp.so
#2  0x00007fffebbc1954 in ros::PollManager::threadFunc() ()
   from /opt/ros/indigo/lib/libroscpp.so
#3  0x00007fffea28ba4a in ?? ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#4  0x00007ffff643e182 in start_thread (arg=0x7fffe4565700)
    at pthread_create.c:312
#5  0x00007ffff616b47d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) 
```

garaemon, 2015-09-14:
実装が適当で、認識結果が空だったりするとsegvで死ぬ気もする

YuOhara, 2015-09-14:
なるほど、
/ocs/env_server/outputの前に、何か先にINPUTしてないといけない感じでしょうか。シミュレータなので、
足りないトピックがあるかもしれないです。

garaemon, 2015-09-14:
env_serverには確か平面認識結果とかが入っているはずで、env_server/outputはrvizのSimpleOccupancyGridで可視化できるはずです。

YuOhara, 2015-09-14:
rviz は　こんな感じでした
[image]

実機とのdiffは

* laser点群が出ないのでとりあえずdepthカメラの点群を使った
* multisenseのimuがないので/imuをそのまま使った

などがあります。

garaemon, 2015-09-14:
認識結果のframe_idがodom_on_groundだったかgroundになってないといけない、とかあったかも

garaemon, 2015-09-14:
segvの原因はcallbackの中で何かしらのエラーが起きているからだと思うので、print仕込んでいくのが筋道がいいかもしれません

garaemon, 2015-09-14:
> multisenseのimuがないので/imuをそのまま使った

これは問題ないはず、NormalDirectionFilterの中でtfを解決していたかは要確認

> laser点群が出ないのでとりあえずdepthカメラの点群を使った

基本的には問題ないはず, FoVだけが問題

YuOhara, 2015-09-14:
> 認識結果のframe_idがodom_on_groundだったかgroundになってないといけない、とかあったかも

大本の/laser_downsampler/outputをmultisense/left/image_rect_colorに変えてしまっているので、
それはあるかもしれないです。

YuOhara, 2015-09-14:
> segvの原因はcallbackの中で何かしらのエラーが起きているからだと思うので、print仕込んでいくのが筋道がいいかもしれません

odom相対にしてみて無理だったら、printして出力を見てみます

garaemon, 2015-09-14:
うまく動いていれば、eusのpickviewが出るんだけど、立ち上がったら即死んでるパターンにも見える。

YoheiKakiuchi, 2015-09-14:
レーザー出ないんだっけ？
最新だと出るようになったと思うので、確認してみて下さい。

YuOhara, 2015-09-14:
#27 まで手元で最新にしてなかったです。
明日確認します

YuOhara, 2015-09-14:
https://github.com/start-jsk/rtmros_choreonoid/pull/57
でした

YuOhara, 2015-09-15:
点群をlaserに変えたら動きました。

シミュレータをあげると、プログラム全体が重すぎて動かないのでこれからPC二台にして不整地試してみます。

2015年9月15日 0:29 Yu Ohara <ohara@jsk.imi.i.u-tokyo.ac.jp>:

> #27 まで手元で最新にしてなかったです。
> 明日確認します
>
>
>> レーザー出ないんだっけ？
>> 最新だと出るようになったと思うので、確認してみて下さい。
>>
>>
>
>
> JSK in the University of Tokyo
> Yu Ohara
> ohara@jsk.imi.i.u-tokyo.ac.jp
> +81-3-5841-7416

garaemon, 2015-09-15:
> シミュレータをあげると、プログラム全体が重すぎて動かないのでこれからPC二台にして不整地試してみます。

どうしても重い場合は、新しいプランナに乗り換えると負荷がかなり減っていると思いますが、判断が難しいところ.

YuOhara, 2015-09-15:
一応htopの出力の一ページ目をあげるとこんな感じでした
[image]

garaemon, 2015-09-15:
うーん、まぁこんなものだろうなぁ。`/joint_states`ってどれくらいの周期(実際の時間で)でているのかな?

YuOhara, 2015-09-15:
rostopic hzだと400hzくらいでした
```
leus@leus-All-Series:~$ rostopic hz /joint_states
subscribed to [/joint_states]
average rate: 433.962
    min: 0.000s max: 0.017s std dev: 0.00386s window: 93
average rate: 397.490
    min: 0.000s max: 0.019s std dev: 0.00452s window: 191
average rate: 402.837
    min: 0.000s max: 0.026s std dev: 0.00478s window: 285
average rate: 398.119
    min: 0.000s max: 0.026s std dev: 0.00481s window: 382
^Caverage rate: 390.492
    min: 0.000s max: 0.026s std dev: 0.00488s window: 461
```

garaemon, 2015-09-15:
Publisherが一人かを調べて欲しいけど、一人だけならちょっと速すぎる気もする。
SeqStateRosBridgeが早く回りすぎ??
100Hzあれば十分だと思っています

garaemon, 2015-09-15:
あとrostopic hzは/use_sim_timeがtrueだとそれに引っ張られるけどchoreonoidはどうなっているんだろうか

YoheiKakiuchi, 2015-09-15:
SeqStateRosBridgeの周期は100Hzにしているつもり。
これは、現実時間の100Hzで回るんじゃないかと思っている。

use_sim_timeはtrueで、SeqStateROSBridgeがclockを出している（hrpsys-simulatorと同じ)になっている。

garaemon, 2015-09-15:
ちなみにrospyの時刻は

https://github.com/ros/ros_comm/blob/indigo-devel/clients/rospy/src/rospy/rostime.py#L210
がコアで、`_rostime_current`があった場合はこれが変える。

`_rostime_current`は初期値Noneで`/clock`のコールバックで代入される
https://github.com/ros/ros_comm/blob/indigo-devel/clients/rospy/src/rospy/rostime.py#L188

garaemon, 2015-09-15:
eusが100%になってるのもちょっと気になりますね

YuOhara, 2015-09-15:
joint_statesのpublisherはひとりでした
```
leus@leus-All-Series:~$ rostopic info /joint_states 
Type: sensor_msgs/JointState

Publishers: 
 * /HrpsysSeqStateROSBridge (http://leus-All-Series:50582/)

Subscribers: 
 * /hrpsys_state_publisher (http://leus-All-Series:51682/)
 * /footcoords (http://leus-All-Series:50785/)
 * /multisense_laser (http://leus-All-Series:50628/)
 * /locomotion/robot_bbox (http://leus-All-Series:60144/)
 * /fc_executive (http://leus-All-Series:46089/)
 * /footstep_controller (http://leus-All-Series:45224/)
 * /lookaround_ground (http://leus-All-Series:38176/)
 * /ocs_executive (http://leus-All-Series:35255/)
 * /request_ik_from_marker (http://leus-All-Series:47553/)
 * /walk_to_object (http://leus-All-Series:55580/)

```

garaemon, 2015-09-15:
400Hzはなんかおかしい気がするけど、realtime factor的なものが0.25なら期待通りなのかな?

YoheiKakiuchi, 2015-09-15:
手元の環境では、以下の表示が出ているのと、
[ INFO] [1442327457.602515159, 20.160000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 100[Hz] (exec 100 Hz, dropped 90)

rostopic hz /joint_statesは 113 くらいなので、リアルタイムファクタの問題だと思います。

YuOhara, 2015-09-16:
まとめて報告しますが（このブランチで報告していいのか微妙ですが）、
まず、PCを二台にわけても、だんだんと重くなってしまいました。

[image]

https://github.com/start-jsk/rtmros_choreonoid/pull/59 がシミュレータの側で上がってるので、これが重いかもしれないです。

YuOhara, 2015-09-16:
ただ、laser系のノードをあげなくてもだんだん重くなっていったので、
choreonoidがだんだん重くなっていっているようにも思えます。

garaemon, 2015-09-16:
> choreonoidがだんだん重くなっていっているようにも思えます。

choreonoidって枚ステップのシミュレーション結果を保存してたりしない? 時間が巻き戻せる的な。
昔openhrpのシミュレータはその機能のせいでだんだんシミュレーションが重くなるという現象があった気がする

YoheiKakiuchi, 2015-09-16:
だんだん重くなるがどれぐらいか分かるかな？
x分たつと、どれぐらい遅くなる的な。

YuOhara, 2015-09-16:
一時間たつとrealtimefactorが0.16くらいになっている気がします。

YuOhara, 2015-09-16:
よく考えたら、
@terasawa @YutaKojio 
が一時間を超えてもスラスラと動かせているので、パソコンの性能かもしれません。

YuOhara, 2015-09-17:
今試してみていて、斜めなところに立った瞬間に、重くなっている気がしました。

YuOhara, 2015-10-16:
不整地上でシミュレータが重いのは，modelを簡略化することで解決しました．

そもそもの，footstep_plannerが死ぬ問題に関しては，
* plannerのinputが空に近い時に起こる様子

@YutaKojio　がいろいろ試してくれたのですが，解決策としては，jvrc的には
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/cfg/EnvironmentPlaneModeling.cfg#L22
を0にして，inputが空にならないようにする．でした．

closed 2015-12-09 by garaemon