jsk-ros-pkg/jsk_control #468: footstep_planningの、足裏の抜ける部分

YuOhara, 2015-09-18:
[image]

pcl_rosの内容かもしれませんが、ロボットの足裏のあたりの平面が延長されて、
その下の本当の平面に足裏があわせられないのですが、どうすればよいでしょうか？

garaemon, 2015-09-18:
https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_footstep_planner/launch/JAXON_RED_footprint.launch

これを良い感じにしてください

YutaKojio, 2015-09-18:
空中に平面が出てしまう問題は上にあるパラメータをいじることで治りました。
しかし、```odom_cloud```から```laser_self_filter_points```の変換で、
実際に足がつけそうな場所まで抜けてしまいます。
[image]

garaemon, 2015-09-18:
> しかし、odom_cloudからlaser_self_filter_pointsの変換で、
> 実際に足がつけそうな場所まで抜けてしまいます。

どういうことだろう?

YutaKojio, 2015-09-18:
１つ前の画像は，白色点群が```odom_cloud```で，```laser_self_filter_points```の出力がピンク色点群なのですが，
laser_self_filter処理後にロボット自身以外の平面まで消えてしまっているため，画像において一歩前に降りるplanningがでなくなってしまいます．
まだコードの中身を読んでいないのですが，
laser_self_filterの処理がlook groundの姿勢ではなくreset-poseの姿勢に対して行われているためではないかと思ったのですが，そういう理解であっていますでしょうか？

garaemon, 2015-09-18:
laser_self_filterと呼んでるのはpr2_navigation_self_filter? それともrobot-bbox.l?

どちらにせよ、姿勢はきちんと考慮されています。
robot-bbox.lの方は、最近pr2_navigation_self_filterでcolladaサポートしたので、こっちに置き換えられる。
もしbboxの方を使ってたらこちらは直方体近似しているのでもしかしたら何か悪さをしているかも。

シミュレーションなので、ちょっとずるいけどパディングパラメータは0にしちゃってよい。

YuOhara, 2015-09-18:
使ってるのは、attention clipperの方です。

点群処理が遅いので、そのノードに届く際に姿勢が変わってしまっている可能性を先ほどは考えてみました。
急いでいたので省略してしまいましたが、実際にはコードの中を読んで、Syncしているか、あるいは別にsubscribeしてるかで挙動が変わる気がするので、そこはちゃんと見たほうが良さそうです。

garaemon, 2015-09-18:
どの点群使ってるか忘れてるんだけど、pr2_navigation_self_filterがかかってる点群を使えれば、attention
clipper周りは消せますね。
こっちの方が信頼できるかも。

処理が重いのならうまくpassthroughのパラメータを調整してみるのかな。

YuOhara, 2015-09-28:
will be solved via https://github.com/jsk-ros-pkg/jsk_demos/pull/1065

closed 2015-09-28 by YuOhara