jsk-ros-pkg/jsk_control #469: :set-auto-balancer-paramを呼ぶと両足をそろえてしまう

YutaKojio, 2015-09-24:
両足のx方向位置が揃っていない状態で
https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_footstep_controller/euslisp/footstep-controller-old.l#L761-L762
を実行すると両足を無理やり揃えようとしてしまいます．

現状，
https://drive.google.com/a/jsk.imi.i.u-tokyo.ac.jp/file/d/0B9Cu2eewhmiwZkNmUHB2blRWZm8/view?usp=sharing
のように，段差を登る前に両足を揃えてしまい，場合によってはそのせいで転けてしまいます．

https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_footstep_controller/euslisp/footstep-controller-old.l#L761-L762
をコメントアウトした状態では，
https://drive.google.com/a/jsk.imi.i.u-tokyo.ac.jp/file/d/0B9Cu2eewhmiweWFMa0VDeXZRVVU/view?usp=sharing
のように両足を揃えることなく登ることができました．

これは，最近起こるようになったことでしょうか？それとも前からあってなにか対策済みでしょうか？

snozawa, 2015-09-24:
もう少し小さいバグ再現コードを作ることは可能かな？
具体的には、

hrpsys-simualtor上で:set-foot-stepsをして足を開いてから:set-auto-balancer-paramをする

くらいの量。

eisoku9618, 2015-09-24:
```bash
tmlaunch hrpsys_ros_bridge_tutorials samplerobot.launch RUN_RVIZ:=false
```

と

```lisp
(progn
  (load "package://hrpsys_ros_bridge_tutorials/euslisp/samplerobot-interface.l")
  (setq *robot* (samplerobot-init))
  (send *ri* :angle-vector (send *robot* :reset-pose))
  (send *ri* :wait-interpolation)
  (send *ri* :start-auto-balancer)
  (send *ri* :set-foot-steps
        (list (list (make-coords :coords (send *robot* :lleg :end-coords :copy-worldcoords) :name :lleg))
              (list (make-coords :coords (send (send *robot* :rleg :end-coords :copy-worldcoords) :translate #f(50 0 0)) :name :rleg))
              ))
  (send *ri* :set-auto-balancer-param :default-zmp-offsets (list #f(0 0 0) #f(0 0 0) #f(0 0 0) #f(0 0 0))))
```
で再現しました．

犯人は僕で，
leg_namesの補間をauto-balancerが入っている状態で出来るようにしたつもりの

https://github.com/eisoku9618/hrpsys-base/commit/cf889ecef1cd1d3410820dbc21941cab647daef8

のコミットが悪さをしていて，
ひとまず以下のdiffで治るかと思います．

```cpp
diff --git a/rtc/AutoBalancer/AutoBalancer.cpp b/rtc/AutoBalancer/AutoBalancer.cpp
index ae73807..687b1b7 100644
--- a/rtc/AutoBalancer/AutoBalancer.cpp
+++ b/rtc/AutoBalancer/AutoBalancer.cpp
@@ -1365,20 +1365,9 @@ bool AutoBalancer::setAutoBalancerParam(const OpenHRP::AutoBalancerService::Auto
                                                                           i_param.graspless_manip_reference_trans_rot[2],
                                                                           i_param.graspless_manip_reference_trans_rot[3]).normalized().toRotationMatrix()); // rtc: (x, y, z, w) but eigen: (w, x, y, z)
   transition_time = i_param.transition_time;
-  if (leg_names_interpolator->isEmpty()) {
-      leg_names.clear();
-      for (size_t i = 0; i < i_param.leg_names.length(); i++) {
-          leg_names.push_back(std::string(i_param.leg_names[i]));
-      }
-      if (control_mode == MODE_ABC) {
-          double tmp_ratio = 0.0;
-          leg_names_interpolator->set(&tmp_ratio);
-          tmp_ratio = 1.0;
-          leg_names_interpolator->go(&tmp_ratio, 5.0, true);
-          control_mode = MODE_SYNC_TO_ABC;
-      }
-  } else {
-      std::cerr << "[" << m_profile.instance_name << "]   leg_names cannot be set because interpolating." << std::endl;
+  leg_names.clear();
+  for (size_t i = 0; i < i_param.leg_names.length(); i++) {
+      leg_names.push_back(std::string(i_param.leg_names[i]));
   }
   pos_ik_thre = i_param.pos_ik_thre;
   rot_ik_thre = i_param.rot_ik_thre;
```

YutaKojio, 2015-09-24:
> hrpsys-simualtor上で:set-foot-stepsをして足を開いてから:set-auto-balancer-paramをする

choreonoid上で:set-foot-stepsをして足を開いてから:set-auto-balancer-paramをする
は，確認済みでそれでも再現していました．

> ひとまず以下のdiffで治るかと思います．

のdiffを入れて試してみます．

snozawa, 2015-09-24:
@eisoku9618 くんがとても良い例を示してくれていますが，
> choreonoid上で:set-foot-stepsをして足を開いてから:set-auto-balancer-paramをする
> は，確認済みでそれでも再現していました．

の部分は，
https://github.com/jsk-ros-pkg/jsk_control/issues/469#issuecomment-142836128
のように，短く再現できる方法・プログラムをかけるととても良いと思います．
自分でデバッグしようとしたときにも重要です．

YutaKojio, 2015-09-24:
> のように，短く再現できる方法・プログラムをかけるととても良いと思います．
> 自分でデバッグしようとしたときにも重要です．

はい，issueを見る方もそちらのほうが見やすいと思うので，
issueを送るときにも短く再現できるプログラムをつけて報告するように心がけます．

上のdiffを入れると足を揃えることはなくなりました．
ありがとうございました．

closed 2015-09-24 by YutaKojio