jsk-ros-pkg/jsk_control #518: footstep_visualizer.pyのzmpの座標が間違っている気がする

garaemon, 2015-11-21:
jaxonだけまちがっているかも

garaemon, 2015-11-21:
@YutaKojio 
CPが支持多角形の外に出てるのって正しい?

[image]

garaemon, 2015-11-21:
rvizでも同様の位置に見えます

YutaKojio, 2015-11-21:
重心速度が０の時はCPの値は重心位置と等しくなります。
なので、もしロボットが静止している状態で出ているのであれば座標変換がおかしい気がします。

garaemon, 2015-11-22:
@YutaKojio 膝を曲げているような状態で静止していて、ただ手が少し物に当たっているような姿勢でした。
hrpsysが出しているCPがおかしいのかもしれませｎ

closed 2015-11-22 by garaemon

garaemon, 2015-11-24:
@YutaKojio 
なんかおかしい?
4足歩行時

[image]

YutaKojio, 2015-11-24:
> 膝を曲げているような状態で静止していて、ただ手が少し物に当たっているような姿勢でした。

については，set-ref-forceをしている状態でそもそも重心が支持多角形を出ていたらしいのでそれでおかしくはなさそうでした．

> 4足歩行時

については，ちゃんと４足のときにも使えるようになっているか確認します．

eisoku9618, 2015-11-25:
RootLinkが地面と水平でない場合に表示の方だけうまく行かないようなプログラムになっていたようでした．

CPが可視化されていると，ZMPとCPが可視化されていると，静止時に実際の重心とモデルの重心の差分が見えてとても便利なので，PR出してもらえると助かります > @YutaKojio

YutaKojio, 2015-11-25:
遅くなりました．
https://github.com/fkanehiro/hrpsys-base/pull/888 で修正しました．

commit 5abe7bc8 (2015-11-22): Merge pull request #520 from garaemon/fix-zmp-location

[jsk_footstep_controller/footstep_visualizer] Fix zmp location by taking     into account z componrnt.     closes #518
commit efaac2b2 (2015-11-22): [jsk_footstep_controller/footstep_visualizer] Fix zmp location by taking
into account z componrnt.

closes #518