jsk-ros-pkg/jsk_control #546: [jsk_footstep_planner] pcl/PointCloudConcatenateDataSynchronizer seems to cause tf2::InvalidArgumentException

k-kimura, 2016-01-19:
### Issue
`roslaunch jsk_footstep_planner heightmap.launch` causes `tf2::InvalidArgumentException`.
I checked nodelets in heightmap_nodelet as follows and I reproduced this error when robot_center_pointcloud_with_map_origin_points is commented in.
1. comment out all nodelets in the heightmap_nodelets
2. comment in nodelets one-by-one from robot_center_pointcloud, which input is /distance_filtered_laser_snapshots/output_cloud
It seems the same error as https://github.com/jsk-ros-pkg/jsk_control/issues/424
/body_on_odom, /map and /odom_init are published as /tf.
Both /robot_center_pointcloud/output and /robot_center_map_origin_points/output have /body_on_odom as their frame_id in headers.

### Environment
hrp2007v with multisenseSL (firmware version 0x0303)

jsk_recognition at master in 810ba37c4fb36bc9d2fafd4329d7cb4d86c1a87d
jsk_common at master in 1a0ecef4643ec4b4a238b95d05f0f516584a510c
jsk_control at master in 79f4aab41059b57f01d8e0fb1eb4fa424044c72d
ros-hydro-pcl-ros 1.1.11-0precise-20150111-0931-+0000 from apt
heightmap.launch is modified to use heightmap_nodelet (not to be combined with multisense_laser)

### How to reproduce
```
hrp2007v$ roslaunch jsk_hrp2_ros_bridge hrp2007.launch # heightmap.launch is disabled by if="false" option
hrp2007v$ roslaunch jsk_footstep_planner heightmap.launch DEBUG_VIEW:=false
```

### gdb output
```
GNU gdb (Ubuntu/Linaro 7.4-2012.04-0ubuntu2.1) 7.4-2012.04
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
For bug reporting instructions, please see:
<http://bugs.launchpad.net/gdb-linaro/>...
Reading symbols from /home/hrpuser/ros/hydro/devel/lib/jsk_topic_tools/standalone_complexed_nodelet...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/hrpuser/ros/hydro/devel/lib/jsk_topic_tools/standalone_complexed_nodelet __name:=heightmap_nodelet __log:=/home/hrpuser/.ros/log/6af1531e-bd01-11e5-b186-00031d0c7c63/heightmap_nodelet-1.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffef540700 (LWP 3174)]
[New Thread 0x7fffeed3f700 (LWP 3175)]
[New Thread 0x7fffee53e700 (LWP 3176)]
[New Thread 0x7fffedd3d700 (LWP 3181)]
[New Thread 0x7fffecf0d700 (LWP 3182)]
[New Thread 0x7fffdffff700 (LWP 3183)]
[New Thread 0x7fffdf7fe700 (LWP 3184)]
[New Thread 0x7fffdeffd700 (LWP 3185)]
[New Thread 0x7fffde7fc700 (LWP 3186)]
[ INFO] [1453199740.830532137]: [int main(int, char**)] i_nodelet 0
[ INFO] [1453199740.830672328]: [int main(int, char**)] remapping: /map_origin_polygon/input => /robot_center_pointcloud/output
[New Thread 0x7fffaa746700 (LWP 5940)]
[ INFO] [1453199766.792916718]: [int main(int, char**)] Succeeded to load nodelet [/map_origin_polygon -- jsk_pcl/StaticPolygonArrayPublisher]
[ INFO] [1453199766.792969914]: [int main(int, char**)] i_nodelet 1
[ INFO] [1453199766.793057491]: [int main(int, char**)] remapping: /map_origin_polygon_sampler/input/polygons => /map_origin_polygon/output_polygons
[ INFO] [1453199766.793124759]: [int main(int, char**)] remapping: /map_origin_polygon_sampler/input/coefficients => /map_origin_polygon/output_coefficients
[New Thread 0x7fffa9f45700 (LWP 5981)]
[ INFO] [1453199766.863401978]: [int main(int, char**)] Succeeded to load nodelet [/map_origin_polygon_sampler -- jsk_pcl/PolygonPointsSampler]
[ INFO] [1453199766.863531440]: [int main(int, char**)] i_nodelet 2
[ INFO] [1453199766.863674949]: [int main(int, char**)] remapping: /robot_center_map_origin_points/input => /map_origin_polygon_sampler/output_xyz
[ INFO] [1453199766.873487696]: [static tf::TransformListener* jsk_recognition_utils::TfListenerSingleton::getInstance()] instantiating tf::TransformListener
[New Thread 0x7fffa9744700 (LWP 6025)]
[ INFO] [1453199766.892528617]: [int main(int, char**)] Succeeded to load nodelet [/robot_center_map_origin_points -- jsk_pcl/TfTransformCloud]
[ INFO] [1453199766.892946323]: [int main(int, char**)] i_nodelet 3
[ INFO] [1453199766.893285148]: [int main(int, char**)] remapping: /robot_center_pointcloud/input => /distance_filtered_laser_snapshots/output_cloud
[ INFO] [1453199766.913616336]: [int main(int, char**)] Succeeded to load nodelet [/robot_center_pointcloud -- jsk_pcl/TfTransformCloud]
[ INFO] [1453199766.913741362]: [int main(int, char**)] i_nodelet 4
[New Thread 0x7fffa8a78700 (LWP 6075)]
[ INFO] [1453199767.562053474]: [onInit] Subscribing to 2 user given topics as inputs:
[ INFO] [1453199767.562124104]:  - /robot_center_pointcloud/output
[... 37 lines cut ...]
---Type <return> to continue, or q <return> to quit---
#13 0x0000000000000000 in ?? ()
(gdb) info thread
  Id   Target Id         Frame
  14   Thread 0x7fffa8a78700 (LWP 6075) "standalone_comp" 0x00007ffff71f00fe in pthread_cond_timedwait@@GLIBC_2.3.2 ()
   from /lib/x86_64-linux-gnu/libpthread.so.0
  13   Thread 0x7fffa9744700 (LWP 6025) "standalone_comp" 0x00007ffff71f00fe in pthread_cond_timedwait@@GLIBC_2.3.2 ()
   from /lib/x86_64-linux-gnu/libpthread.so.0
  12   Thread 0x7fffa9f45700 (LWP 5981) "standalone_comp" 0x00007ffff71f00fe in pthread_cond_timedwait@@GLIBC_2.3.2 ()
   from /lib/x86_64-linux-gnu/libpthread.so.0
  11   Thread 0x7fffaa746700 (LWP 5940) "standalone_comp" 0x00007ffff71f00fe in pthread_cond_timedwait@@GLIBC_2.3.2 ()
   from /lib/x86_64-linux-gnu/libpthread.so.0
  10   Thread 0x7fffde7fc700 (LWP 3186) "standalone_comp" 0x00007ffff71efd84 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib/x86_64-linux-gnu/libpthread.so.0
  9    Thread 0x7fffdeffd700 (LWP 3185) "standalone_comp" 0x00007ffff71efd84 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib/x86_64-linux-gnu/libpthread.so.0
  8    Thread 0x7fffdf7fe700 (LWP 3184) "standalone_comp" 0x00007ffff71efd84 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib/x86_64-linux-gnu/libpthread.so.0
* 7    Thread 0x7fffdffff700 (LWP 3183) "standalone_comp" 0x00007ffff69450d5 in raise () from /lib/x86_64-linux-gnu/libc.so.6
  6    Thread 0x7fffecf0d700 (LWP 3182) "standalone_comp" 0x00007ffff71efd84 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib/x86_64-linux-gnu/libpthread.so.0
---Type <return> to continue, or q <return> to quit---
  5    Thread 0x7fffedd3d700 (LWP 3181) "standalone_comp" 0x00007ffff71f00fe in pthread_cond_timedwait@@GLIBC_2.3.2 ()
   from /lib/x86_64-linux-gnu/libpthread.so.0
  4    Thread 0x7fffee53e700 (LWP 3176) "standalone_comp" 0x00007ffff71efd84 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib/x86_64-linux-gnu/libpthread.so.0
  3    Thread 0x7fffeed3f700 (LWP 3175) "standalone_comp" 0x00007ffff69fbc23 in select () from /lib/x86_64-linux-gnu/libc.so.6
  2    Thread 0x7fffef540700 (LWP 3174) "standalone_comp" 0x00007ffff71f29fa in __lll_unlock_wake () from /lib/x86_64-linux-gnu/libpthread.so.0
  1    Thread 0x7ffff7fbe780 (LWP 2985) "standalone_comp" 0x00007ffff71f00fe in pthread_cond_timedwait@@GLIBC_2.3.2 ()
   from /lib/x86_64-linux-gnu/libpthread.so.0

k-kimura, 2016-01-19:
How did you solve this problem? > @aginika

k-okada, 2016-01-20:
if you solve this issue offline, please describe how did you solve this.

aginika, 2016-01-21:
@k-kimura I didn't get any error when I try to do it in gdb. So I don't know the solution sorry.

k-kimura, 2016-01-21:
@aginika thanks.
> if you solve this issue offline, please describe how did you solve this.

I found this error issue while checking https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2_multisense_local.launch#L135-#L138 with @orikuma san.
But we have not solved this problem yet...

YoheiKakiuchi, 2016-09-06:
Ping @k-kimura

k-kimura, 2016-09-06:
When I checked jsk_footstep_planner's heightmap.launch on mensa (hrp2 remote desktop PC) just now, it seems no error about ```tf2::InvalidArgumentException```.

### Environment
- Launch https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2_multisense_local.launch on mensa (hrp2 remote desktop PC)
- Add ```if="false"``` option at https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_robot_common/jsk_robot_startup/launch/multisense_local.launch#L140
- jsk_footstep_planner at master in 37af46f0d15dad6225d6f7f88d38b37525a271a6
- jsk_robot_startup at master in 9a5cf76bed15d96b9acf815d9f51aff7f59e9611
- jsk_hrp2_ros_bridge at master in 6fe5369a89d8c52fa01086025c754f9d055db166

### Execution result of heightmap.launch
```
[http://hrp2007v:10007][192.168.96.242] jskuser@mensa:~$ roslaunch jsk_footstep_planner heightmap.launch
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
... logging to /home/jskuser/.ros/log/a7f025f8-7410-11e6-b186-00031d0c7c63/roslaunch-mensa-25574.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/jskuser/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.96.242:56626/

SUMMARY
========

PARAMETERS
 * /accumulated_heightmap/center_frame_id
 * /accumulated_heightmap/fixed_frame_id
 * /accumulated_heightmap_pointcloud_static/target_frame_id
 * /heightmap_nodelet/nodelets_0
 * /latest_heightmap/max_x
 * /latest_heightmap/max_y
 * /latest_heightmap/min_x
 * /latest_heightmap/min_y
 * /latest_heightmap/resolution_x
 * /latest_heightmap/resolution_y
 * /map_origin_polygon/frame_ids
 * /map_origin_polygon/polygon_array
 * /map_origin_polygon/use_message
 * /robot_bbox_clipper/frame_id
 * /robot_bbox_clipper/use_multiple_attention
 * /robot_bounding_box/analysis_level
 * /robot_bounding_box/height_offset
 * /robot_bounding_box/padding_x
 * /robot_bounding_box/padding_y
 * /robot_bounding_box/padding_z
 * /robot_center_map_origin_points/target_frame_id
 * /robot_center_pointcloud/target_frame_id
 * /robot_center_pointcloud_bbox_clipped/negative
 * /robot_center_pointcloud_with_map_origin_points/input_topics
 * /robot_center_pointcloud_with_map_origin_points/max_queue_size
 * /robot_center_pointcloud_with_map_origin_points/output_frame
 * /rosdistro
 * /rosversion

NODES
  /heightmap_visualization/latest/
    colorize_heightmap (jsk_perception/colorize_float_image)
    heightmap_view (image_view/image_view)
  /heightmap_visualization/accumulated/
    colorize_heightmap (jsk_perception/colorize_float_image)
    heightmap_view (image_view/image_view)
  /heightmap_visualization/filtered/
    colorize_heightmap (jsk_perception/colorize_float_image)
    heightmap_view (image_view/image_view)
  /
    heightmap_nodelet (jsk_topic_tools/standalone_complexed_nodelet)
    robot_bounding_box (jsk_footstep_controller/robot-boundingbox.l)

ROS_MASTER_URI=http://hrp2007v:10007

[... 70 lines cut ...]
process[heightmap_visualization/accumulated/heightmap_view-8]: started with pid [28423]
[ INFO] [1473163598.653913216]: [main] done reading parmaeters
[ INFO] [1473163598.654012871]: [main] loaded nodelets: 12
[ INFO] [1473163598.654057925]: [main] loaded nodelet: /accumulated_heightmap
[ INFO] [1473163598.654093037]: [main] loaded nodelet: /accumulated_heightmap_pointcloud
[ INFO] [1473163598.654126760]: [main] loaded nodelet: /accumulated_heightmap_pointcloud_static
[ INFO] [1473163598.654161466]: [main] loaded nodelet: /latest_complete_heightmap
[ INFO] [1473163598.654199119]: [main] loaded nodelet: /latest_heightmap
[ INFO] [1473163598.654228931]: [main] loaded nodelet: /map_origin_polygon
[ INFO] [1473163598.654260954]: [main] loaded nodelet: /map_origin_polygon_sampler
[ INFO] [1473163598.654292996]: [main] loaded nodelet: /robot_bbox_clipper
[ INFO] [1473163598.654327668]: [main] loaded nodelet: /robot_center_map_origin_points
[ INFO] [1473163598.654361544]: [main] loaded nodelet: /robot_center_pointcloud
[ INFO] [1473163598.654391561]: [main] loaded nodelet: /robot_center_pointcloud_bbox_clipped
[ INFO] [1473163598.654420886]: [main] loaded nodelet: /robot_center_pointcloud_with_map_origin_points
[ WARN] [1473163602.319324916]: [jsk_topic_tools::ConnectionBasedNodelet::warnNeverSubscribedCallback] '/accumulated_heightmap' subscribes topics only with child subscribers.
[ WARN] [1473163602.488793864]: [jsk_topic_tools::ConnectionBasedNodelet::warnNeverSubscribedCallback] '/accumulated_heightmap_pointcloud_static' subscribes topics only with child subscribers.
```

Then, the following topics about heightmap_visualization are published correctly.
```
/heightmap_visualization/latest/colorize_heightmap/output
/heightmap_visualization/filtered/colorize_heightmap/output
/heightmap_visualization/accumulated/colorize_heightmap/output
```