jsk-ros-pkg/jsk_control #572: .cfg files need to be fixed and released

dirk-thomas, 2016-03-22:
The .cfg files in this repo need fixing (similar as jsk-ros-pkg/jsk_common#1359): e.g. https://github.com/jsk-ros-pkg/jsk_control/blob/be2468326eba643b858f73e4b50aab5c369f90de/jsk_footstep_planner/cfg/FootstepPlanner.cfg#L6-L12

The fixes also need to be released. Currently the Indigo jobs fail since the old code falls back to the non-catkin code generation which won't generate the header in the correct location.

k-okada, 2016-03-23:
Sorry for this, fixed in -> https://github.com/jsk-ros-pkg/jsk_control/pull/573

k-okada, 2016-03-23:
merged in https://github.com/ros/rosdistro/pull/10837, 
I think I have done what I can do right now, rest of failing package depends on naoqi_driver,  https://github.com/ros-naoqi/libqi-release/issues/3

closed 2016-03-23 by k-okada