jsk-ros-pkg/jsk_control #624: HRP2/JAXON その他のjsk_footstep_controller/footcoords.cppの出すodomの名前を変更する

YoheiKakiuchi, 2016-09-06:
humanoidsからodomとして出ているものの実態がAutobalancerの出しているodomであるので、
Autobalancerの出しているodomをabc_odom (名前は議論があっていいと思う）として、
odom = abc_odom とするのがいいのではないか。

理由としては、Autobalancer以外のodomが出てきたときの対応と、
footstep-controller等でodomをAutobalancerの原点として使っているためわかりにくいという点が挙げられます。

yoshimalucky, 2016-11-09:
これはトピック名の話でしょうか？tfのframe_idの話でしょうか？

関連したことで質問ですが、odomのトピックとtfはそれぞれ同等のものとして対応しているのでしょうか。

odom_initからodomを経由して/BODYまでのtfはfootcoordsから出ている一方で、
odomのトピック(odomからBODY)はHrpsysSeqStateBridgeから直接出ているので。

それと、ドキュメントを読みましたが、
odom
odom_on_ground 
odom_init
の違いがよくわかっていません。

odom_on_ground は HRP2の画像を見れば説明と一致していますが、
JAXONのgazeboのシミュレーションのでは、
odomもodom_on_groundと同じく床に張り付いています。

odom_initはロボットを歩かせても特に更新されず、
odomに張り付いたままです。

ちなみに、leg_odomは最後に両足が設置したところにセットされるような挙動を示しています。

実験環境はgazebo4のJAXONです。よろしくお願いします。

YoheiKakiuchi, 2016-11-22:
footcoordsの出しているトピックで必須のものはどれだろうか？

メモ

~~~
トピック

Publications: 
 * /footcoords/odom_init_transform [geometry_msgs/TransformStamped]
 * /footcoords/odom_init_pose_stamped [geometry_msgs/PoseStamped]
 * /footcoords/leg_odometry [geometry_msgs/PoseStamped]

 * /footcoords/base_vel [geometry_msgs/TwistStamped]
 * /footcoords/state [std_msgs/String]
 * /footcoords/low_level_info [jsk_footstep_controller/FootCoordsLowLevelInfo]
 * /footcoords/contact_state [jsk_footstep_controller/GroundContactState]

 * /footcoords/debug/lfoot_pose [geometry_msgs/Pose]
 * /footcoords/debug/rfoot_pose [geometry_msgs/Pose]
 * /tf [tf2_msgs/TFMessage]

Subscriptions: 
 * /tf [tf2_msgs/TFMessage]
 * /tf_static [tf2_msgs/TFMessage]

 * /joint_states [sensor_msgs/JointState]

 * /odom [nav_msgs/Odometry]
 * /imu [sensor_msgs/Imu]
 * /zmp [geometry_msgs/PointStamped]

 * /odom_init_trigger [std_msgs/Empty]

 * /rfsensor [geometry_msgs/WrenchStamped]
 * /lfsensor [geometry_msgs/WrenchStamped]

 * /floor_coeffs [unknown type]

出している tf

odom_init -> odom
odom -> odom_on_ground
odom -> BODY
BODY -> ground
BODY -> body_on_odom
BODY -> zmp
BODY -> leg_odom
~~~