jsk-ros-pkg/jsk_control #684: solve-qpoases-qpのint-status

mmurooka, 2017-06-28:
@snozawaさん
qpoasesがときどきinternal errorになって解けないときに，
変数を変換する(適当な正則行列Aによってy=Axでxからyに変換する)と解けることがあるというのをオフラインでお話させていただきましたが，

これを効率よくやるために，qpが解けなかった時にそれがinternal errorなのかinfeasibleなのかを区別したくて，
https://github.com/jsk-ros-pkg/jsk_control/blob/master/eus_qpoases/euslisp/eus-qpoases.l#L191
の`int-status`を`solve-qpoases-qp`関数の外から見れるようにしたいです．
返り値を変えるとこれまでのプログラムが動かなくなってしまうのが問題ですが，何か良い案はありますでしょうか．
あまり綺麗ではないですがグローバル変数にセットする，くらいしかないものでしょうか．

snozawa, 2017-06-29:
これはirteusのIKでも同様のことを思ったことがあり、決め手がなかったです。
（ikのloop回数などを外から知りたい、など）

上記のときにでてた同様の解決法で、思い出すと以下があった気がします。

1. （クラスにして）クラスのslotsかなにかにdebug関係のものを追加し、そこに値を保存しておく
2. グローバル変数にする
3. 引数でdebug関係のものを与えるようにし、関数のなかでそれを破壊的にかきかえる。ログをとりたいユーザは明示的にそれを引数で与えて、中身をみる

```
(defun test (&key (debug-log (list :log1 nil :log2 nil)))
  (setf (cadr (memq :log1 debug-log)) 1.0)
  (setf (cadr (memq :log2 debug-log)) 5.0)
  t)

(let ((debug-log (list :log1 nil :log2 nil)))
  (test :debug-log debug-log)
  (print debug-log)
  t)
```

おそらく、１が罪悪感が少ない素直な感じになると思います。
3も当時の誰かの案で、実装は簡単で、かつおもしろいなと思った気がします。

３の亜種で、実際にこの関数限定で適用できる方法は、実はやってる例はあります。
https://github.com/jsk-ros-pkg/euslib/blob/master/demo/nozawa/multicontact_motion_planner/qp-inverse-kinematics2.l#L106-L114
のように、@mmurookaさんのいってる関数にはqp-solverをlamdaで与えられますが、
それでなんとか引数をもらってきてると思います。

mmurooka, 2017-06-29:
見逃していて上のコメントを見て気づいたのですが，
そもそも引数の`:status`でqpがどのように成功/失敗したかは取ってくることができたんですね（＝3の方法がすでにやられていたんですね）．
なのでやりたかったことはできそうでした．

closed 2017-06-29 by mmurooka