jsk-ros-pkg/jsk_control #689: jsk_footstep_plannerでマーカーを動かすと死ぬ

mttamtam, 2017-10-24:
@YoheiKakiuchi 

rtmlaunch hrpsys_choreonoid_tutorials jaxon_red_choreonoid.launch LOAD_OBJECTS:=true ENVIRONMENT_YAML:=$(rospack find hrpsys_choreonoid_tutorials)/config/terrain.yaml
roslaunch hrpsys_choreonoid_tutorials jaxon_multisense_local.launch
を上げた状態で、
下記のようにフットステッププランナを立ち上げ、ゴール位置のマーカーを動かそうとするとその瞬間にrvizが死にます。

tamura@aconcagua:~/ros/indigo/src/jsk-ros-pkg/jsk_control/jsk_footstep_planner (master)$ roslaunch jsk_footstep_planner JAXON_RED_footstep_planner_perception.launch 
... logging to /home/tamura/.ros/log/f76d7c64-b8b8-11e7-85d3-bc5ff4ea565e/roslaunch-aconcagua-20301.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://aconcagua:43194/

SUMMARY
========

CLEAR PARAMETERS
 * /joy_manager/

PARAMETERS
 * /footstep_marker/default_footstep_margin: 0.2
 * /footstep_marker/disable_tf: False
 * /footstep_marker/footstep_size_x: 0.237418
 * /footstep_marker/footstep_size_y: 0.134017
 * /footstep_marker/frame_id: map
 * /footstep_marker/lleg_footstep_offset: [0.015502, 0.0100...
 * /footstep_marker/rleg_footstep_offset: [0.015502, -0.010...
 * /footstep_marker/use_default_step_as_goal: True
 * /footstep_planner/collision_bbox_offset: [0, 0, 0.8]
 * /footstep_planner/collision_bbox_size: [0.35, 0.65, 0.65]
 * /footstep_planner/default_lfoot_to_rfoot_offset: [0, -0.2, 0]
 * /footstep_planner/footstep_size_x: 0.26
 * /footstep_planner/footstep_size_y: 0.135
 * /footstep_planner/lleg_footstep_offset: [0.015, 0.0, 0.0]
 * /footstep_planner/local_move_theta: 0.1
 * /footstep_planner/local_move_theta_num: 0
 * /footstep_planner/local_move_theta_offset: 0.0
 * /footstep_planner/local_move_x: 0.02
 * /footstep_planner/local_move_x_num: 1
 * /footstep_planner/local_move_x_offset: 0.14
 * /footstep_planner/local_move_y: 0.01
 * /footstep_planner/local_move_y_num: 0
 * /footstep_planner/local_move_y_offset: 0.0
 * /footstep_planner/obstacle_resolution: 0.1
 * /footstep_planner/plane_estimation_max_iterations: 200
 * /footstep_planner/plane_estimation_min_inliers: 100
 * /footstep_planner/plane_estimation_min_ratio_of_inliers: 0.8
 * /footstep_planner/plane_estimation_normal_distance_weight: 0.2
 * /footstep_planner/plane_estimation_normal_opening_angle: 0.2
 * /footstep_planner/plane_estimation_outlier_threshold: 0.024
 * /footstep_planner/plane_estimation_use_normal: False
 * /footstep_planner/project_goal_state: True
 * /footstep_planner/project_start_state: True
 * /footstep_planner/rleg_footstep_offset: [0.015, 0.0, 0.0]
 * /footstep_planner/skip_cropping: False
 * /footstep_planner/successors: [{'y': 0, 'x': 0,...
 * /footstep_planner/support_check_vertex_neighbor_threshold: 0.02
 * /footstep_planner/support_check_x_sampling: 3
 * /footstep_planner/support_check_y_sampling: 3
 * /footstep_planner/support_padding_x: 0.0
 * /footstep_planner/support_padding_y: 0.0
 * /footstep_planner/transition_limit_pitch: 0.6
 * /footstep_planner/transition_limit_roll: 0.6
 * /footstep_planner/transition_limit_x: 0.5
 * /footstep_planner/transition_limit_y: 0.35
 * /footstep_planner/transition_limit_yaw: 0.4
 * /footstep_planner/transition_limit_z: 0.4
 * /footstep_planner/use_lazy_perception: False
[... 126 lines cut ...]
[ WARN] [1508849319.028062092, 74.749000000]: #<ros::simple-action-client #X11bcfdc8 /thk_hand_left> is not respond, pr2-interface is disabled
[ WARN] [1508849319.280046843, 74.830000000]: ROSEUS is already installed as /jaxon_footstep_controller
[ INFO] [1508849324.929754962, 76.619000000]: simple footstep controller is ready as /jaxon_footstep_controller
[ INFO] [1508849357.002083847, 89.037000000]: pose_up
[ INFO] [1508849357.002120771, 89.037000000]: start planning
[ INFO] [1508849357.004987445, 89.042000000]: planCB
[ INFO] [1508849357.005400394, 89.042000000]: start_solver timeout: 10.000000
[ INFO] [1508849357.005606761, 89.042000000]: Took 0.000080 sec
Perception took 0.000000 sec
Planning took 0.000080 sec
1 path
open list: 0
close list:0
[ INFO] [1508849357.005649258, 89.042000000]: use_obstacle_model: 1
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'Ogre::Exception'
  what():  OGRE EXCEPTION(5:): Could not find font Arial in MovableText::setFontName
[ WARN] [1508849357.099110094, 89.067000000]: [footstep_planner] action-result-cb may recieved old client's goal
[ WARN] [1508849357.099157438, 89.067000000]:      expected goal id nil
[ WARN] [1508849357.099180764, 89.067000000]:      recieved goal id /footstep_marker-3-89.42000000
[rviz-9] process has died [pid 20514, exit code -6, cmd /opt/ros/indigo/lib/rviz/rviz -d /home/tamura/ros/indigo/src/jsk-ros-pkg/jsk_control/jsk_footstep_planner/config/jaxon_footstep_planner_perception.rviz __name:=rviz __log:=/home/tamura/.ros/log/f76d7c64-b8b8-11e7-85d3-bc5ff4ea565e/rviz-9.log].
log file: /home/tamura/.ros/log/f76d7c64-b8b8-11e7-85d3-bc5ff4ea565e/rviz-9*.log

k-okada, 2017-10-24:
使っているjsk_rviz_plugins のバージョンはなんですか？
https://github.com/jsk-ros-pkg/jsk_visualization/pull/676
かな，とおもいました．

mttamtam, 2017-10-26:
遅くなりすいません。
岡田先生のご指摘のとおりで、
```
tamura@aconcagua:~/ros/indigo/src/jsk-ros-pkg/jsk_visualization/jsk_rviz_plugins (master)$ git log
commit 30ee668e47af175057548041802d9846c65918c9
Merge: a1a5ab2 997c218
Author: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Date:   Wed Oct 12 01:39:16 2016 +0900

    Merge pull request #643 from mmurooka/empty-service-panel-unused-varialbe
    
    [jsk_rviz_plugins/src/empty_service_call_interface.cpp ] remove unused variables.
```
だったのでjsk_rviz_pluginsが１年前ぐらいだったのを最新版にしたら問題なく動きました。
ありがとうございました。

closed 2017-10-26 by mttamtam

k-okada, 2017-10-26:
良かったです．
手元にソースツリー持つとこういうことがおこって，じゃあ常にwstool
updateしたらいいかというと，regressionも心配だし，ということで，自分でいじることがなければdebを使っておくのがいいですね．
ということで，リリースしました．週明けには使えると思います．
https://github.com/ros/rosdistro/pull/16173