jsk-ros-pkg/jsk_control #719: How do I add new plug-ins for jsk_teleop_joy?

weiqiyang, 2019-04-16:
I am trying to add some new plug-ins to `jsk_teleop_joy`.
As a test, I copied `jsk_teleop_joy/src/jsk_teleop_joy/plugin/joy_pose_6d.py`, and renamed it as `jyo_pose_6d.py`. I also changed the class name in the file from `JoyPose6D` to `JyoPose6D`.

But when I ran the launch file, it gave an error message as bellow, and `MyPose6D` was not showed in the overlay menu.

> [ERROR] [WallTime: 1555408864.470235] cannot find JyoPose6D in plugins for jsk_teleop_joy

The launch file was as the following.

```
<launch>
  <arg name="CONTROLLER_TYPE" default="auto" />
  <arg name="DEV" default="/dev/input/js0" />
  <arg name="POSE_REMAP" default="joy_pose" />
  <node pkg="joy" type="joy_node" name="joy_driver" output="screen" >
    <param name="dev" type="string" value="$(arg DEV)" />
    <param name="deadzone" value="0.2" />
    <param name="autorepeat_rate" value="40" />
    <param name="coalesce_interval" value="0.025" />
  </node>
  <node pkg="jsk_teleop_joy" type="joy.py" name="joy" output="screen"
        clear_params="true">
    <param name="controller_type" value="$(arg CONTROLLER_TYPE)" />
    <rosparam subst_value="true">
      plugins:
        'Pose6D':
           class: 'JoyPose6D'
           args:
             publish_pose: True
             pose: $(arg POSE_REMAP)
             follow_view: true
        'Verbose':
           class: 'VerboseStatus'
           args:
             publish_pose: True
             pose: $(arg POSE_REMAP)
             follow_view: true
        'MyPose6D':
           class: 'JyoPose6D'
           args:
             publish_pose: True
             pose: $(arg POSE_REMAP)
             follow_view: true
    </rosparam>
  </node>
</launch>
```

What are the additional steps I have to do to add a new plug-in?

k-okada, 2019-04-17:
i think you need to update
https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_teleop_joy/plugin.xml
--
◉ Kei Okada

weiqiyang, 2019-04-18:
It works! Thank you!

closed 2019-04-18 by weiqiyang