jsk-ros-pkg/jsk_control #732: duplicated example.launch

sktometometo, 2019-11-07:
I am trying to execute samples of this package.

And when I tried to launch example.launch, I had got an error.

```
$ roslaunch jsk_footstep_planner example.launch 
multiple files named [example.launch] in package [jsk_footstep_planner]:
- /home/sktometometo/ros/ws_jsk_ros_pkg/src/jsk-ros-pkg/jsk_control/jsk_footstep_planner/launch/example.launch
- /home/sktometometo/ros/ws_jsk_ros_pkg/src/jsk-ros-pkg/jsk_control/jsk_footstep_planner/launch/eusplanner/example.launch
Please specify full path instead
The traceback for the exception was written to the log file

$ diff example.launch eusplanner/example.launch 
$
```

There seems to be two example.launch files in this package and thier name and contens are exactly same.
Is there any special purpose of this file duplication?

k-okada, 2019-11-07:
@orikuma can you help him?

orikuma, 2019-11-07:
Historically, jsk_footstep_planner was developed in euslisp, and it was reimplemented in C++ after DRC.
These examples are for the euslisp version, and they seems to be left unintentionally in launch directory.
Therefore, I think they should be only in the eusplanner directory.
However, the euslisp version is no longer maintained as far as I know.
If you just want to see how the C++ version works, you can run `roslaunch jsk_footstep_planner footstep_planner_perception_sample.launch` instead of these examples.

k-okada, 2019-11-07:
thank you @sktometometo can you update https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_footstep_planner/README.md based on @orikuma comment?

sktometometo, 2019-11-08:
Thank you, @orikuma.
@k-okada , I have created a pullreq about this issue.
Could you please review it?

sktometometo, 2019-11-08:
I have fixed my error.

closed 2019-11-15 by sktometometo