jsk-ros-pkg/jsk_control #747: Question about the usage of unit test codes

rjshim, 2020-05-11:
Hi @garaemon,

I have a question about how to write unit test codes for robot controllers.
Since you do not use a physical robot while unit testing your controller, you can only use a simulator, in this case, Gazebo, as you elaborated in [this code](https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_footstep_controller/test/test_footstep_controller.py), but this is not really following the syntax recommended in the ROS [instruction](http://wiki.ros.org/gtest). 
Instead of just running each command-line on the top and run your test.py executable, is there any other way to check if your controller works fine by just using catkin_make with the run_tests option? 

Any advice would be appreciated. Thank you in advance,
Ryan

garaemon, 2020-05-11:
As far as I remember, [the code](https://github.com/jsk-ros-pkg/jsk_control/blob/master/jsk_footstep_controller/test/test_footstep_controller.py) is a demonstration code rather than a unit test code.

In order to make test_footstep_controller.py a unit test, [rostest](http://wiki.ros.org/rostest) can be a good solution.

rjshim, 2020-05-12:
@garaemon 

I thought it was a test code as it was in the test folder. Thank you for your advice. I will try that

Ryan

closed 2020-05-12 by rjshim