jsk-ros-pkg/jsk_control #748: Optmotinogenの障害物回避IKが上手くいかない

Kanazawanaoaki, 2020-05-24:
Optmotinogenの[sample-arm-reach-ik-obstacle](https://github.com/jsk-ros-pkg/jsk_control/tree/master/eus_qp/optmotiongen#sample-arm-reach-ik-obstacle)に使われている障害物回避IKをPR2で試してみようとした(https://github.com/Kanazawanaoaki/oneweek_kanazawa/issues/1 )ところ以下のような問題が発生しました。

### :collision-avoidance-link-pair を指定する必要がある
sample-arm-reach-ik-obstacleと同じように`:obstacles (list *obstacle*)`として障害物を指定してもPR2の場合は障害物回避をすることが出来ませんでした。
[image]
(https://gist.github.com/Kanazawanaoaki/1810958f83c44c037b9cb286a047b385#file-obstacle-ik-test-01-l )
@pazeshunに`:obstacles (list *obstacle*)`の代わりに`:collision-avoidance-link-pair (mapcar #'(lambda (l) (list l *obstacle*)) (send *robot* :links))`とする必要があることを教えていただきましたが、[README](https://github.com/jsk-ros-pkg/jsk_control/blob/master/eus_qp/optmotiongen/README.md )から辿るにはPR2のregraspの[サンプルのコード](https://github.com/jsk-ros-pkg/jsk_control/blob/master/eus_qp/optmotiongen/euslisp/demo/demo-pr2-regrasp-object.l )を注意深く読む以外にない気がします。  

### 障害物を回避しきれていない
`:collision-avoidance-link-pair`を(https://gist.github.com/Kanazawanaoaki/1810958f83c44c037b9cb286a047b385#file-obstacle-ik-test-02-l )のように指定して実行してみたのですが、障害物を回避しきれていなくて重なってしまっているように見えます。
[image]
`(normal-ik)`として`:inverse-kinematics`を使った時と、`(obstacle-ik)`として`:inverse-kinematics-optmotiongen`を使った時の挙動を比べると障害物を指定できていて回避していようとしているように見えます。完全には回避しきれない状態で解き終わってしまうこともあるのでしょうか。

Naoki-Hiraoka, 2020-05-25:
PR2は特殊なロボットで，`:links`や`:link-list`に全リンクが含まれていないため，それらのリンクの干渉が考慮されないのが原因ではないかと思います．

` (objects (send *robot* :link-list (send *robot* :rarm :end-coords :parent)))`
[image]

`(objects (send *robot* :links))`
[image]

`:collision-avoidance-link-pair`にしっかりリンクを指定してやるとうまく行くのではないかと思いますが，PR2のリンク構造によってはそのままだとうまく動かないかもしれません．

今回の原因と直接関係はありませんが，現在のoptmotiongenが使用しているPQPによる干渉チェックは，物体同士が既に干渉している時に離れる方向を正しく出力しないため，一度干渉すると干渉したままになってしまいやすいという問題があります．これは，https://github.com/euslisp/jskeus/pull/555 がリリースされると改善できるのですが，今のところは初期姿勢で物体と干渉していないように注意する必要があります．

Kanazawanaoaki, 2020-05-25:
ありがとうございます。
本当ですね、、`(objects (send *robot* :link-list (send *robot* :rarm :end-coords :parent)))`を見て驚きました。考慮したいリンクを全部きちんと指定するようにしないといけないみたいですね。  

初期姿勢で物体と干渉していないかのチェックもするようにしようと思います。

Kanazawanaoaki, 2020-05-26:
PR2はぶつけるオプションもあるということでしたが、  
```
:collision-avoidance-link-pair (mapcar #'(lambda (l) (list l *obstacle*)) (append (send *robot* :link-list (send *robot* :rarm :end-coords :parent)) (list (send *robot* :r_forearm_link_lk) (send *robot* :r_upper_arm_link_lk) (send *robot* :r_gripper_palm_link_lk) (send *robot* :r_gripper_r_finger_link_lk) (send *robot* :r_gripper_l_finger_link_lk) (send *robot* :r_gripper_r_finger_tip_link_lk)
```
と考慮したいリンクを`append`することで回避する事ができました。  
[image]
```
(objects (append (send *robot* :link-list (send *robot* :rarm :end-coords :parent)) (list (send *robot* :r_forearm_link_lk) (send *robot* :r_upper_arm_link_lk) (send *robot* :r_gripper_palm_link_lk) (send *robot* :r_gripper_r_finger_link_lk) (send *robot* :r_gripper_l_finger_link_lk) (send *robot* :r_gripper_r_finger_tip_link_lk) (send *robot* :r_gripper_l_finger_tip_link_lk))))
```
[image]

k-okada, 2020-05-26:
メモ：https://github.com/jsk-ros-pkg/jsk_model_tools/issues/171
に議論があるもので．https://github.com/jsk-ros-pkg/jsk_model_tools/issues/171#issuecomment-633789925 に汚いけど，プログラマブルにcollision-linkを正しく（見えるように）する方法が書いてあります．他の人やロボットでもこれが必要な状況が増えたら，各自がコードを書かずに，上流で対応したいともうので，ココにやってきた人は足跡を残していてください．

ちなみにこれはURDFがassocの代わりにfixed jointで別リンクをつなげていく，というのが原因だともいます．

pazeshun, 2020-05-26:
@Kanazawanaoaki 
まだ
https://github.com/jsk-ros-pkg/jsk_control/blob/master/eus_qp/optmotiongen/README.md
に情報が載ってないので、少なくとも、このissueは開けておいた方がよいと思いました。

Kanazawanaoaki, 2020-05-26:
そうですね、開けておくことにします。