jsk-ros-pkg/jsk_control #749: optmotiongen の kxr を表示させるデモでエラーが発生する

hiroya1224, 2020-06-07:
cc. @mmurooka 
optmotiongen で kxr を動かすデモ `demo-kxr-inverse-kinematics-statics.l` を試そうとしているのですが， 
- euslisp を source から入れてインストール
-  jskeus を `mmurooka/jskeus-bullet` から clone, make

をして，
```
git clone -b optmotiongen-experimental-latest https://github.com/mmurooka/jsk_control.git
(branch は optmotiongen-experimental も試しましたが同様のエラーでした)
```
として適宜 rosdep install および catkin build したのち，
```
cd ~/catkin_ws/src/jsk_control/eus_qp/optmotiongen/euslisp/demo
roseus demo-kxr-inverse-kinematics-statics.l
(demo-kxr-inverse-kinematics-statics)
```
とすると，以下のエラーが出て止まってしまいます:
```
/home/leus/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :worldrot in (bt-collision-distance b1 b2)
```

問題の箇所は jsk_control/eus_qp/optmotiongen/euslisp/instant-configuration-task.l の 2517 行目から定義されている :update-collision-inequality-constraint で発生しているようです．

ちなみに同じく demo ディレクトリにある
demo-pr2-regrasp-object.l
demo-fetch-elbow-stretch.l
は正常に動作することを確認しています．

Naoki-Hiraoka, 2020-06-08:
以下の行を削除すると正常に動作しました．
https://github.com/mmurooka/jsk_control/blob/392efb3ba7745f14da1cbe4a4d052f962060b4e7/eus_qp/optmotiongen/euslisp/demo/demo-kxr-inverse-kinematics-statics.l#L84-L86
KXRのリンク名が変わったらしく，torso-topというリンクが存在しないためエラーになっているようです．

また，jskeusのmmurooka/jskeus-bulletを作りなおしたブランチが存在し，それが既にjskeusのmasterにmergeされている(relseaseはされていない)ので，jskeusのmasterをソースから入れてコンパイルしても動きます．
https://github.com/euslisp/jskeus/pull/555

hiroya1224, 2020-06-08:
動作確認しました
ありがとうございます！

closed 2020-06-08 by hiroya1224